DelfiSpace/EPS_FlightSoftware #8: Gas gauge accuracy seems poor

StefanoSperetta, 2020-01-08:


MFvandenBos, 2020-01-09:
[image]

MFvandenBos, 2020-01-09:
Note the battery current in the picture of the previous comment

MFvandenBos, 2020-01-09:
[image]

note how the gas gauge value has changed compared to the previous picture

StefanoSperetta, 2020-01-09:
The current being negative means we are charging. It should not have been red though...

But what's the time difference between the two screenshots?
EPS sw revision? Current master EPS?

I think I removed the code in the gas gauge using the offset but purely outputting the charge accumulator directly...

MFvandenBos, 2020-01-13:
Time difference is ~ 5 minutes, sw version is current master. It discharged from 4916 to 64817

StefanoSperetta, 2020-01-13:
the problem is that the "offset" is still in the LTC2942 code. I tried to implement it correctly but we still have glitches in there...

What I can do is disable this offset but we will anyway see a jump when the charger stops the charging process (it is implemented in hardware). 

So, what will happen is that the battery will keep charging and when the voltage reaches 4.2V (on the protection switch), the charge will jump to 65535 and stay there till the battery voltage starts decreasing. But for the rest it should change linearly with charge/discharge...

MFvandenBos, 2020-02-26:
solved by new LTC chips

closed 2020-02-26 by MFvandenBos