DelfiSpace/EPS_FlightSoftware #9: NMI pin reset is never cleared

MFvandenBos, 2020-01-22:
[image]

As in the above image, this reset cause is not cleared, except by re-uploading code.  This can place the satellite in unrecoverable condition

MFvandenBos, 2020-01-22:
[image]

MFvandenBos, 2020-01-22:
Problem was caused by systick being too fast and the watchdog too slow. [https://github.com/DelfiSpace/EPS/issues/24](url)

StefanoSperetta, 2020-01-22:
Cause identified, Issue needs to be fixed at hardware level.

StefanoSperetta, 2020-01-22:
PeriodicTaskNotifier.cpp still needs to be fixed...

Use again TIMER32 might be a good temporary fix, but it would be best to link the fix of this issue to https://github.com/DelfiSpace/DelfiPQcore/issues/17

CasperBroekhuizen, 2020-01-23:
Workaround for now with the TIMER32. As the SystickPeriod seems to be a 32bit value, but is actually a 24bit value (http://www.ti.com/lit/ug/slau356i/slau356i.pdf). Meaning that a 1 second period (48MCycles) is too big for this register. 

See commit: https://github.com/DelfiSpace/DelfiPQcore/commit/e352f39656140fab495fbd4a70bca4afcf1d6bd8

StefanoSperetta, 2020-02-10:
Workaround implemented, fixing the issue will require quite some time. The issue for fixing this issue on DelfiPQCore is already in place, so closing this one...

closed 2020-02-10 by StefanoSperetta