DelfiSpace/EPS_FlightSoftware #13: EPS is rebooting under discharge

StefanoSperetta, 2020-04-02:
During a discharge test (Telemetry_2_response_0 - Discharge on 26/03/2020) the EPS started rebooting towards the end of the test.

BUS 1 was turning automatically ON and was attached to a 800mA load.

Battery voltage was 3.2 - 3.3V, Bus1 voltage was < 3V, UR bus voltage was 3.4V.
(see [reboot.xlsx](https://github.com/DelfiSpace/EPS_FlightSoftware/files/4424093/reboot.xlsx))

In theory, BUS1 should not turn ON automatically till the battery voltage goes above 3.6V (it might also be an old version of the EPS SW checking for UR_bus voltage but still, this should not be happening).

@meseulu can you check which EPS SW version is currently running (so @CasperBroekhuizen for the command to send on the bus). I want to understand what is happening in this case. We might need to replicate this test (just the part where the battery is almost fully discharged) and attach the console to see what is happening.

One more issue: it can be clearly seen that the telemetry is made by all zeroes (line 34 of the xls sheet, for example) when telemetry is requested just after a reboot. This happens because the telemetry collection task did not finish its first iteration. This is not very nice as we are outputting garbage on purpose.

We could stop responding to telemetry requests if the first iteration of the task did not run, we might also force an iteration of the task straight after initialisation of the code. What do you think about it?

CasperBroekhuizen, 2020-04-03:
For documentation, we found out that the following software was on the EPS during these tests:
 EPS - 0CE0E332 ->  https://github.com/DelfiSpace/EPS_FlightSoftware/commit/CE0E332
 Workspace - 037AB27E   -> https://github.com/DelfiSpace/FlightSoftwareWorkspace/commit/37ab27

StefanoSperetta, 2020-04-03:
This means that the thresholds were applied to the latest telemetry data taken from the unregulated bus  and the thresholds were:

`#define VBUS_LOW_THRESHOLD          3000
#define VBUS_HIGH_THRESHOLD         3200`

The EPS is rebooting so, probably, the battery voltage drops below 2.8V under load. When the load is removed, the battery bounces back to 3.2V meaning that there is a "hysteresis" of at least 0.4V. 

The SW has been modified to monitor the battery voltage directly and the thresholds have been changed to 3.0 - 3.6V.

I would repeat the test using  the latest EPS master code (currently f0282c6293e1a458fd96f37d419853ad745e82c4).